reefangel/Libraries #205: Portal doesn't receive any historical info (inaccurate dosing pump status.. etc.)

lnevo, 2015-01-27:
Would it be possible to maintain a bitmap in a byte or int that would contain the last 5 minutes or 300 seconds of status for each port and this way the portal could more accurately show the status of the ports? 

The justification for this effort would be better visibility and reporting for heaters / dosers / etc that may cycle on/off frequently and be missed by the current reporting.

I have some ideas for tracking the status, but would need the portal to be able to parse it..

robertoimai, 2015-01-27:
The problem I had in the past was that the portal would be flooded with data from fast moving data, such as wavemakers and DC shortwave.
That's why it was locked in 5 min interval.

lnevo, 2015-01-27:
Yeah I'm thinking we track it in code and then have it be part of the 5 minute push.

robertoimai, 2015-01-27:
That would still require every change between the 5 minutes interval to be sent, right?

lnevo, 2015-01-27:
No, the granularity of the portal isn't going to give us per minute view anyway. So the idea is just to track if a port has changed on/off during the 5 minute period. One byte could tell you that for one relay bar.

robertoimai, 2015-01-27:
I don't know if I follow.

lnevo, 2015-01-27:
RHIST=255; // All ports were modified at some point in past 5 minutes

or RHIST=0;

Then if you see activity on a port you can plot it.

You could potentially add 4 total variables per relay box and gain visibility per minute past 4 minutes plus present state.

robertoimai, 2015-01-27:
I wrote this response about 4 times already....

I kept deleting and starting again while thinking because my thoughts kept changing what I had just written... lol

So, after this back and forth, I think you are right. This may just work.

lnevo, 2015-01-28:
The only question in my mind is do we keep it simple and just store the current relay state in RHIST* or do we use it for flags to track activity. 

Flags would be more accurate to catch a momentary on/off but we wouldn't be able to track actual usage and you'd have to extrapolate out the relay states in the portal.

Relay state is the easiest to capture but we'd only capture the top of each minute and easier to keep continuous. I think it would cover most dosing pump scenarios but maybe not temperature based devices because they aren't timed. 

I think I'd just go with the relay state due to ease of implementation.