reefangel/Libraries #208: Broken analogWrite function

robertoimai, 2015-02-07:
The commit https://github.com/reefangel/Libraries/commit/8a0a3e6e783aebf888a06725cb16c7270fec6ebe#commitcomment-9530591 broke the analogWrite function

lnevo, 2015-02-07:
Just curious why is this broken? Wouldn't we want the capability to test and overwride a channel even if using DCPump?

robertoimai, 2015-02-07:
This is the fix for the patch you mentioned that was broken before.
You can still override, but the channels won't be overriden to memory location when lightson is triggered.
When not using DCPump class, PWM channels get overriden to memory locations automatically when the lightson mode is triggered.

lnevo, 2015-02-07:
Ok good I can test both of those things then when I attempt it.

commit 99c578f1 (2015-02-07): Fixed #define for PWM override on lightson mode
close #208
closed 2016-01-05 by robertoimai