reefangel/Libraries #212: Missing Math.h in Globals.cpp

curtbinder, 2015-03-11:
I just updated my local Dev branch environment to match these libraries and I got an error trying to compile. The error was a missing "Math.h" file from Globals.cpp
I looked and could not find the Math.h library or file anywhere. My only guess is that it was meant to be "math.h". I changed that on my dev branch and it compiles just fine for me (or at least it seems to be just fine).

Is there supposed to be a Math.h file or Math library file included in the libraries? Or is it actually supposed to be the standard, included math.h file?
I created this issue to discuss it and correct if it is incorrect or maybe I just need to update my setup.

robertoimai, 2015-03-11:
I think this is because of the way linux and windows treat filenames.

In windows, it is case insensitive, so that's probably why we never had compile issues.

But we should change Globals.cpp to match the same case as the file itself.

It is part of the core libraries:

hardware\tools\avr\avr\include\math.h

curtbinder, 2015-03-11:
Ahh

curtbinder, 2015-03-11:
Makes sense... I was wondering why nobody encountered it before.

closed 2015-08-03 by robertoimai