reefangel/Libraries #224: PWM Override channels broken for the apps

curtbinder, 2016-01-24:
In a commit, a102ca864c2024d24c39da5114b49e1ef7977483 on July 1, 2015, the PWM override global defines were altered and moved around. When that happened, that broke all the PWM overrides in the android app (and probably others). It shifted the overrides down 2 by moving the actinic2 and daylight2 defines up above to be inline with the actinic and daylight defines.

So, that leaves version 1.1.2 incompatible with the current release of my android app (not a big deal as it can be fixed). The 2 options I can see are:  
1. Change the defines back to the way they were and everything is fine onward.
2. Leave it as it stands and then modify the app(s) to handle versions prior to 1.1.2 a certain way and from 1.1.2 onward handle it differently. 

If we did option 2, I would just put in an advanced setting to use pwm overrides for versions before 1.1.2 and have it be the default checked. This would be similar to what we did for libraries version 0.8.5 when there was a change with the relay boxes (or something along those lines).

If we did option 1, it would be easy for me because I would not have anything to do. :)

I'm impartial to the decision. Either option will be fine with me. I'm mainly bringing this up for people to be aware of it in case they were not (as I was because I have not kept up on the changes being made).

curtbinder, 2016-01-24:
This does not affect the standard Actinic and Daylight channels OR the 16 channel expansion. It only affects the PWM Expansion, AI expansion and Radion Expansion.

robertoimai, 2016-01-24:
I'm going to revert back and change the u-app and star board stuff. Easier that way.

curtbinder, 2016-01-25:
ok. like I said, I'm ok with whatever you choose to do. I was just pointing it out.

commit 3769fad0 (2016-01-26): Reverted back PWM override IDs. It broke the Android App.
close #224
closed 2016-01-26 by robertoimai