reefangel/Libraries #255: I2C bus is getting locked by the touch screen

robertoimai, 2019-11-20:
Reported here:
https://forum.reefangel.com/viewtopic.php?f=12&t=6967
https://forum.reefangel.com/viewtopic.php?f=2&t=6963
There seems to be a bug causing the bus to be locked by the touch screen that goes away when the touch screen is physically touched.

curtbinder, 2019-11-21:
very interesting. maybe this is the issue that I was having with my star and the controlling of my heater. especially since I know mythtv port is good as well as the heater.

curtbinder, 2019-11-21:
I'll add this hear instead of the forum. I had it happen last night. My screen had blanked after a water change (water change mode) like normal. I noticed later in the evening that my multi channel water level expansion and salinity expansion modules were not responding (a 6.0 on salinity and 0 for all the 4 channels of the water level). It's also possible that the temperature was not reading properly but I cannot confirm it and will have to wait to see if the history was reported on the portal/uapp. I was able to query the controller for the status using my android app and getting the /r99 status. This morning, I tapped the screen and within a few seconds the salinity and water level modules were working/reporting the proper status again.

closed 2020-01-25 by robertoimai